athom-tech/athom-configs #47: Smart plug v2 with latest esphome config produces errors in the HA logs

sefininio, 2024-03-15:
2024-03-15 06:16:12.636 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.pool_pump_smart_plug_v2_wifi_strength (<class 'homeassistant.components.esphome.sensor.EsphomeSensor'>) is using native unit of measurement '%' which is not a valid unit for the device class ('signal_strength') it is using; expected one of ['dBm', 'dB']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+esphome%22

sefininio, 2024-03-24:
Hello
Any updates on this?

hellcry37, 2024-04-13:
I am beginning to think that Athom Tech just do some products that look cool on paper but when it comes to software updates they are 0 ...

bucky2076, 2024-04-20:
also getting this error in the logs...

the wifi strength is defined as % when it should be something else...

2024-04-19 19:21:52.461 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.athom_smart_plug_v2_a788db_wifi_strength (<class 'homeassistant.components.esphome.sensor.EsphomeSensor'>) is using native unit of measurement '%' which is not a valid unit for the device class ('signal_strength') it is using; expected one of ['dBm', 'dB']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+esphome%22