athom-tech/athom-configs #50: Smart plug v2 with latest esphome config spams logbook with Uptime reports

Adrian57, 2024-04-08:
Every time the uptime sensor updates (60 seconds), a line is logged to the logbook. If you have a lot of smartplugs, its a lot of entries!

hellcry37, 2024-04-13:
I am beginning to think that Athom Tech just do some products that look cool on paper but when it comes to software updates they are 0 ...

lloydhazlett, 2024-04-14:
I noticed the same thing, from this commit merged by @tarontop:
https://github.com/athom-tech/athom-configs/commit/192ea652d37afdbeea634751795009b3af2d6183

> I am beginning to think that Athom Tech just do some products that look cool on paper but when it comes to software updates they are 0

@hellcry37 In this case it looks like it was a community contribution (good!), doesn't seem like the author @Roving-Ronin has any formal association with Athom.

Roving-Ronin, 2024-04-15:
To confirm I do NOT have any formal links to Athom. Hell, be good if I was on their books, having just bricked another of my own plugs, testing updates for others to benefit from. As with most things ESPHome or Home Assistant related, the enhancement and/or bugfixing is something reliant upon the users to assist and contribute.  

The preferable option as Lloyd kindly pointed out, is to assist / contribute... I'm no programmer myself but read the documentation and learn how to do it and lend up hand as and where you can, its more rewarding also than tossing 'grading remarks'.

As per above 'mention' Pull 53 created to provide you the easy option to change the update frequency in the substitutions section.
See this line:  occassional_update_sensor_interval: 3min

Roving-Ronin, 2024-04-16:
@Adrian57 

To cut down on these entries being logged, try adding to configuration yaml:
```
recorder: !include recorder.yaml
```

then making a recorder.yaml file in the /homeassistant directory, with:

```
#purge_keep_days: 10
exclude:
  domains:
    - automation
    - update
    - button
    - light
    - script
  entity_globs:
    - sensor.weather_*
    - sensor.daylight_*
    - sensor.days_*
    - sensor.*_device_uptime
    - sensor.*_project_name
    - sensor.*_project_version
    - sensor.*_wifi_ip_address
    - sensor.*_wifi_ip_address
    - sensor.*_wifi_ssid
    - sensor.*_wifi_strength
    - sensor.*_wifi_signal
    - sensor.*_mac_address
  entities:
    - sensor.date_time
    - sensor.internet_time
    - sensor.time
```

then reload your configuration (or restart/reboot etc). 
Note '#purge_keep_days: 10' is hashed out, as 10 days is the default timeframe for the recorder to keep data.