athom-tech/athom-configs #59: Presence Sensors not installing ESPHome 2024.5.1

fifthecho, 2024-05-20:
When trying to install the latest ESPHome, I get the following message:

`ERROR Error binary size: Error: ESP does not have enough space to store OTA file. Please try flashing a minimal firmware (remove everything except ota)`

I'm using a super-basic config:

```
substitutions:
  name: athom-presence-sensor-<MAC>
  friendly_name: athom-presence-<LOCATION>
packages:
  Athom_Technology.Presence_Sensor: github://athom-tech/athom-configs/athom-presence-sensor.yaml
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: <KEY>
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

```

jpettitt, 2024-05-20:
The binary size jumped from 529k to 660k between 2024.5.0 and 2024.5.1

**My Workaround**

1. Downgrade to 5.4.0 (should be in system -> backups in HA)
1. If 5.4 installs cleanly you're good to go.  
1. If it still complains about not enough space try rebooting the sensor before the update.

1. If that fails open the config and comment out the encryption in the `encryption` stanza in the config.

1. Rebuild and install, it should now be small enough.  
1. Once it reboots the 5.4 code uncomment the encryption and install again.   

This has worked for my sensors.

HaraldGithub, 2024-05-21:
> At present, the fastest and most convenient solution is to download the firmware locally to the computer, and then use your own data cable to flash the firmware directly. You only need to connect the data cable.

Very good idea, but where could I download the ESP-Firmware?
At this page ( https://github.com/athom-tech/athom-configs/releases/tag/2023.8.3  ) only a FW from Sep2023 is available.

nickmcgill, 2024-05-21:
Within ESPHome, click on the three dots of the device to update, select install, then manual download.

Mika255, 2024-05-22:
> We may need to add
> 
> ```
>  platformio_options:
>     board_build.ldscript: eagle.flash.2m.ld
> ```
> 
> The chip is 2MB, while the default configuration is 1MB
> 
> example
> 
> ```
> esphome:
>   name: "${name}"
>   friendly_name: "${friendly_name}"
>   name_add_mac_suffix: true
>   project:
>     name: "${project_name}"
>     version: "${project_version}"
>     platformio_options:
>        board_build.ldscript: eagle.flash.2m.ld
> ```

did this work for you? It didn't help. It's still showing 1mb

```
packages:
  Athom_Technology.Presence_Sensor: github://athom-tech/athom-configs/athom-presence-sensor.yaml
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
  platformio_options:
    board_build.ldscript: eagle.flash.2m.ld
```

nickmcgill, 2024-05-22:
I did this in two steps:
Wirelessly flashed
`substitutions:
  name: athom-presence-sensor-xxxxxx
  friendly_name: Athom presence sensor
#packages:
#  Athom_Technology.Presence_Sensor: github://athom-tech/athom-configs/athom-presence-sensor.yaml
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
  platformio_options:
    board_build.ldscript: eagle.flash.2m.ld
#Added
esp8266:
  board: esp8285
ota:
logger:
#end
api:
  encryption:
    key: xxxxxxxxx=

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
`
then a second wireless flash with
`substitutions:
  name: athom-presence-sensor-xxxxxx
  friendly_name: Athom presence sensor
packages:
  Athom_Technology.Presence_Sensor: github://athom-tech/athom-configs/athom-presence-sensor.yaml
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
  platformio_options:
    board_build.ldscript: eagle.flash.2m.ld
#esp8266:
#  board: esp8285
#ota:
#logger:
api:
  encryption:
    key: xxxxxxxxx=

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
`

I couldn't use a one-step method for wireless updates. I'm assuming that the original firmware was for a 1Mb chip, so therefore wouldn't permit the update. I could be wrong!

nickmcgill, 2024-05-22:
I had to do this in two steps.
First ota update with

> `substitutions:
  name: athom-presence-sensor-xxxxxx
  friendly_name: Athom presence sensor
#packages:
#  Athom_Technology.Presence_Sensor: github://athom-tech/athom-configs/athom-presence-sensor.yaml
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
  platformio_options:
    board_build.ldscript: eagle.flash.2m.ld

esp8266:
  board: esp8285
ota:
logger:

api:
  encryption:
    key: xxxxxxxxxxxxxxxx=

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
`
Second ota update:

> `substitutions:
  name: athom-presence-sensor-xxxxxx
  friendly_name: Athom presence sensor
packages:
  Athom_Technology.Presence_Sensor: github://athom-tech/athom-configs/athom-presence-sensor.yaml
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
  platformio_options:
    board_build.ldscript: eagle.flash.2m.ld

#esp8266:
#  board: esp8285
#ota:
#logger:

api:
  encryption:
    key: xxxxxxxxxxxxxxxx=

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
`
This works for me. I couldn't update in one go. I assume that the existing firmware was compiled for a 1 Mb chip so rejected the modified upgrade. I could be wrong!

HaraldGithub, 2024-05-22:
> This works for me.

Sorry for asking, but could you explain how I've to do this steps:
- where have I to copy the code in
- how could I start the OTA process
- what is important 

Thank you very much!

nickmcgill, 2024-05-22:
I use ESPHome within HomeAssistant, so I can only help you from that.
HA/ESPHome should have detected and adopted the presence sensor.
From the ESPHome addon (in HA) you will see the ESPHome devices (mine is still called Athom Presence Sensor).
On the bottom of the device card, there is "visit, edit, logs". Click on "edit", edit the code, then top right of the editor is "install"

I hope that makes sense.

HaraldGithub, 2024-05-22:
Yes, that make sense and I'm using also HA.

I'll try and report here!

nickmcgill, 2024-05-22:
I had that initially, then I commented out the "packages" entries and added 
```
esp8266:
  board: esp8285
ota:
logger:
```
and installed it wirelessly - that's the "minimal" firmware that worked for me.

2nd step was to uncomment the "packages" entries and comment out my additions, then install that.

You will  need to set the encryption key to your setting not "xxxxxx" (similarly with the device name)

HaraldGithub, 2024-05-22:
Only to make it clear:

Frist I should use this yaml (replaced with my specific key):

`
substitutions:
  name: athom-presence-sensor-xxxxxx
  friendly_name: Athom presence sensor
#packages:
#  Athom_Technology.Presence_Sensor: github://athom-tech/athom-configs/athom-presence-sensor.yaml
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
#  platformio_options:
#    board_build.ldscript: eagle.flash.2m.ld

esp8266:
  board: esp8285
ota:
logger:

api:
  encryption:
    key: xxxxxxxxxxxxxxxx=

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
`

than second installation with:

`
substitutions:
  name: athom-presence-sensor-xxxxxx
  friendly_name: Athom presence sensor
#packages:
#  Athom_Technology.Presence_Sensor: github://athom-tech/athom-configs/athom-presence-sensor.yaml
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
  platformio_options:
    board_build.ldscript: eagle.flash.2m.ld

#esp8266:
#  board: esp8285
#ota:
#logger:

api:
  encryption:
    key: xxxxxxxxxxxxxxxx=

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
`
right?

nickmcgill, 2024-05-22:
On the first, do not comment out
```
  platformio_options:
    board_build.ldscript: eagle.flash.2m.ld
```
because that is what is needed to access more memory (I think).
On the second installation you need to uncomment the "packages" section

You seem to have quotes round the # marks for some reason - they should not be there.
# is a comment, uncomment by deleting the #

HTH

HaraldGithub, 2024-05-22:
Okay than like this:

first:

```
substitutions:
name: athom-presence-sensor-xxxxxx
friendly_name: Athom presence sensor
#packages:
# Athom_Technology.Presence_Sensor: github://athom-tech/athom-configs/athom-presence-sensor.yaml
esphome:
name: ${name}
name_add_mac_suffix: false
friendly_name: ${friendly_name}
 platformio_options:
 board_build.ldscript: eagle.flash.2m.ld

esp8266:
board: esp8285
ota:
logger:

api:
encryption:
key: xxxxxxxxxxxxxxxx=

wifi:
ssid: !secret wifi_ssid
password: !secret wifi_password
```
second:

```
substitutions:
name: athom-presence-sensor-xxxxxx
friendly_name: Athom presence sensor
#packages:
# Athom_Technology.Presence_Sensor: github://athom-tech/athom-configs/athom-presence-sensor.yaml
esphome:
name: ${name}
name_add_mac_suffix: false
friendly_name: ${friendly_name}
# platformio_options:
# board_build.ldscript: eagle.flash.2m.ld

#esp8266:
#board: esp8285
#ota:
#logger:

api:
encryption:
key: xxxxxxxxxxxxxxxx=

wifi:
ssid: !secret wifi_ssid
password: !secret wifi_password
```

If I remember right, that's exact the code of my first attempt. But I'll try again.

nickmcgill, 2024-05-22:
No, second attempt uncomment the packages  and platformio sections 
```
substitutions:
name: athom-presence-sensor-xxxxxx
friendly_name: Athom presence sensor
packages:
  Athom_Technology.Presence_Sensor: github://athom-tech/athom-configs/athom-presence-sensor.yaml
esphome:
name: ${name}
name_add_mac_suffix: false
friendly_name: ${friendly_name}
 platformio_options:
   board_build.ldscript: eagle.flash.2m.ld

#esp8266:
#board: esp8285
#ota:
#logger:

api:
encryption:
key: xxxxxxxxxxxxxxxx=

wifi:
ssid: !secret wifi_ssid
password: !secret wifi_password
```

HaraldGithub, 2024-05-22:
Thank you, now it work's for me!

nickmcgill, 2024-05-22:
> Thank you, now it work's for me!

Yay!

jpettitt, 2024-05-22:
So I did a build using the 2 step method above and it worked, however any attempt to send another build fails with this error

```
INFO Connecting to 192.168.203.70
INFO Uploading /data/build/athom-presence-sensor-9cee44/.pioenvs/athom-presence-sensor-9cee44/firmware.bin (612256 bytes)
INFO Compressed to 414762 bytes
ERROR Error binary size: Unknown error from ESP
```

Unless I reboot the device first.

I also noticed that it's still saying I obnly have 1M of flash despite the `board_build.ldscript: eagle.flash.2m.ld`

```
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.4
|-- Wire @ 1.0
|-- ArduinoJson @ 6.18.5
RAM:   [=====     ]  49.1% (used 40256 bytes from 81920 bytes)
Flash: [======    ]  58.2% (used 608105 bytes from 1044464 bytes)
```

What am I missing?

Config I'm using:
```yaml
substitutions:
  name: athom-presence-sensor-9cee44
  friendly_name: Athom Presence Dining Room
packages:
  Athom_Technology.Presence_Sensor: github://athom-tech/athom-configs/athom-presence-sensor.yaml
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
  platformio_options:
    board_build.ldscript: eagle.flash.2m.ld

api:
  encryption:
    key: xxxxxxxxxxxx

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

nickmcgill, 2024-05-23:
I'm getting the same reported 1M flash, and initially got errors on subsequent install attempts.
It now installs every time. 
I've no idea what's going on.

propi62, 2024-05-23:
for me its not installing. i will wait some time and hope it will install one day.

[image]

GVDub, 2024-05-25:
Looks like this worked for me. Guess we'll know for sure at the next ESPHome update, right? 

BTW, don't forget to backup your code before editing and validate your new code before trying to install. Great way to make sure that any copy/paste functions didn't hose you indentations.

nickmcgill, 2024-05-25:
My update went through without any problems.
"Update all" from the ESPHome addon worked first time.

HaraldGithub, 2024-05-25:
How look's your yaml-code before you click "Update all"?

Like this:

```
substitutions:
  name: athom-presence-sensor-xxxxxx
  friendly_name: Athom presence sensor
packages:
  Athom_Technology.Presence_Sensor: github://athom-tech/athom-configs/athom-presence-sensor.yaml
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
#  platformio_options:
#    board_build.ldscript: eagle.flash.2m.ld

#esp8266:
#  board: esp8285
#ota:
#logger:

api:
  encryption:
    key: xxxxxxxxxxxxxxxx=

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

nickmcgill, 2024-05-25:
No, you need to uncomment the following lines
```
#  platformio_options:
#    board_build.ldscript: eagle.flash.2m.ld
```
so they are 
```
  platformio_options:
    board_build.ldscript: eagle.flash.2m.ld
```
and keep them like that for all subsequent builds otherwise it will revert to the 1M setting (I believe)

HaraldGithub, 2024-05-25:
Okay, thank you, but I think I've had this like you replay - so we'll see at next Update of ESPhome. ;-)

GVDub, 2024-05-28:
Just a data point. OTA updates for 2024.5.4 worked flawlessly for me with the revised YAML code.

fear, 2024-08-22:
i had the same problem updating to version 2024.8.0. The solution worked for me but i had to add the ota plattform, otherwise it wouldnt compile

```
substitutions:
  name: presence-sensor-bad
  friendly_name: presence_bad
packages:
  Athom_Technology.Presence_Sensor: github://athom-tech/athom-configs/athom-presence-sensor.yaml
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
  platformio_options:
    board_build.ldscript: eagle.flash.2m.ld

#esp8266:
#  board: esp8285
#ota:
#  platform: esphome
#logger:

api:
  encryption:
    key: mykey

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: .my.home

```

RyanWor, 2024-09-24:
> I had to do this in two steps. First ota update with
> 
> ```
> substitutions:
>   name: athom-presence-sensor-xxxxxx
>   friendly_name: Athom presence sensor
> #packages:
> #  Athom_Technology.Presence_Sensor: github://athom-tech/athom-configs/athom-presence-sensor.yaml
> esphome:
>   name: ${name}
>   name_add_mac_suffix: false
>   friendly_name: ${friendly_name}
>   platformio_options:
>     board_build.ldscript: eagle.flash.2m.ld
> 
> esp8266:
>   board: esp8285
> ota:
> logger:
> 
> api:
>   encryption:
>     key: xxxxxxxxxxxxxxxx=
> 
> 
> wifi:
>   ssid: !secret wifi_ssid
>   password: !secret wifi_password
> ```
> 
> Second ota update:
> 
> ```
> substitutions:
>   name: athom-presence-sensor-xxxxxx
>   friendly_name: Athom presence sensor
> packages:
>   Athom_Technology.Presence_Sensor: github://athom-tech/athom-configs/athom-presence-sensor.yaml
> esphome:
>   name: ${name}
>   name_add_mac_suffix: false
>   friendly_name: ${friendly_name}
>   platformio_options:
>     board_build.ldscript: eagle.flash.2m.ld
> 
> #esp8266:
> #  board: esp8285
> #ota:
> #logger:
> 
> api:
>   encryption:
>     key: xxxxxxxxxxxxxxxx=
> 
> 
> wifi:
>   ssid: !secret wifi_ssid
>   password: !secret wifi_password
> ```
> 
> This works for me. I couldn't update in one go. I assume that the existing firmware was compiled for a 1 Mb chip so rejected the modified upgrade. I could be wrong!
> 
> Sorry about the multiple edits, struggling with formatting (rtfm!)

This solution worked perfectly for me as well. Thank you!

hevel86, 2024-10-03:
> i had the same problem updating to version 2024.8.0. The solution worked for me but i had to add the ota plattform, otherwise it wouldnt compile
> 
> ## 1st Flash
> ```yaml
> substitutions:
>   name: athom-presence-<MAC>
>   friendly_name: athom-presence-<LOCATION>
> 
> #packages:
> #  Athom_Technology.Presence_Sensor: github://athom-tech/athom-configs/athom-presence-sensor.yaml
> esphome:
>   name: ${name}
>   name_add_mac_suffix: false
>   friendly_name: ${friendly_name}
>   platformio_options:
>     board_build.ldscript: eagle.flash.2m.ld
> 
> esp8266:
>   board: esp8285
> ota:
>   platform: esphome
> logger:
> 
> api:
>   encryption:
>     key: mykey
> 
> wifi:
>   ssid: !secret wifi_ssid
>   password: !secret wifi_password
>   domain: .my.home
> ```
> 
> ## 2nd flash
> ```yaml
> substitutions:
>   name: athom-presence-<MAC>
>   friendly_name: athom-presence-<LOCATION>
> packages:
>   Athom_Technology.Presence_Sensor: github://athom-tech/athom-configs/athom-presence-sensor.yaml
> esphome:
>   name: ${name}
>   name_add_mac_suffix: false
>   friendly_name: ${friendly_name}
>   platformio_options:
>     board_build.ldscript: eagle.flash.2m.ld
> 
> #esp8266:
> #  board: esp8285
> #ota:
> #  platform: esphome
> #logger:
> 
> api:
>   encryption:
>     key: mykey
> 
> 
> wifi:
>   ssid: !secret wifi_ssid
>   password: !secret wifi_password
>   domain: .my.home
> ```

This still did not work for me.  I had to flash the stock firmware first, and then it allowed me to flash the latest.  However, I am *still* having to flash the stock firmware each time I want to update.

marcosikkens, 2024-10-17:
I did an update with the usb cable which worked perfectly a couple of months ago. I thought after that the new updates would work perfectly again. But the next time the normal ota update would fail with the same error as above. 

I am still confused as why this happens. Does the size of the package only fail on the ota update proces? 

Would be nice if this was actually fixed!

marcosikkens, 2024-11-30:
So i tried the way hevel86 mentioned and it does indeed work for a single update.

NateroniPizza, 2025-02-09:
Just bought some of these, and am running into the same thing. I read somewhere else that these are being discontinued - is that the case?

marcosikkens, 2025-02-09:
Where did you read this? I only knew about the name change.

NateroniPizza, 2025-02-11:
> > Just bought some of these, and am running into the same thing. I read somewhere else that these are being discontinued - is that the case?
> 
> No discontinuation

> > Where did you read this? I only knew about the name change.
> 
> ESP32C3 (PS01C3) is used in the EU. A small amount of stock in non-EU regions uses ESP8285. Once the stock is cleared, all will use ESP32C3 version

Got it - thank you for the claritification. It was just a single person mentioning it, so they might've heard about the board being replaced and took that to mean discontinuation.

Is there a drop-in replacement board available that will allow us to flash updates over the air?

marcosikkens, 2025-02-11:
@NateroniPizza this issue is open for a long time. No fixes have been made there is only a workaround.

Your guess is a good as mine, but i dont think this will ever be fixed. M

marcosikkens, 2025-06-07:
Ah remove it from the config yaml in esphome. For  it would still give me an error about the size when uploading. Maybe i have to flash a minimal config first again.