athom-tech/athom-configs #61: Blue LED turns on again on pir motion detection

propi62, 2024-05-22:
Hello
At night in bedroom the blue LED is to bright.
The light literally blinds me and I can even see a beam of light.
Is there a way to permanent disable the LED other then gluetape or destroy the led?

nickmcgill, 2024-05-22:
Go to the on-board web server and switch off the status LED
It works for me ...

pgale, 2024-05-23:
> Go to the on-board web server and switch off the status LED It works for me ...
> 
> [edit] ... and then it switches on again

Yes, this only turns the LED off temporarily until it's triggered again. The permanent fix is discussed above by tarontop. I do this on some of my sensors and it works perfectly. This is also an old issue that has been discussed before on this forum (I think)

closed 2024-06-11 by propi62