athom-tech/athom-configs #62: ESPHome 2024.5.0 ruined my Key Switches

iandarbey, 2024-05-24:
Hi,

As the title says any key switch units I've updated now intermittently connect and disconnect from HA.
The web page won't work to do a manual install and downgrade the firmware.

Not sure what my next step is?

Roving-Ronin, 2024-05-24:
@iandarbey check around.... https://github.com/esphome/issues/issues/5826

Known java issue, wait for 2024.5.3 that should have webserver v3.2.1 that fixes this. 
You can't access webserver locally on 'non ESP32' devices atm, but ESPHome when working properly should be able to recompile a new version with web_server 3.2.1 and reflash your switch, at which point you'll be able to access it locally.

In the meantime, your switch should still be readable via its existing HA connection.

Roving-Ronin, 2024-07-01:
@iandarbey This issue should have been resolved. Suggest closing the issue?