athom-tech/athom-configs #74: New power restore mode defaults to off leading to unexpected power off.

jpettitt, 2024-09-03:
You added power restore mode to smart plugs.   Unfortunately when I ran the ESP home update this defaulted to "off" and switched off my plugs unexpectedly.   It would have been better if it defaulted to "last state".

jpettitt, 2024-09-04:
I saw you added a description but didn't change the value, why not set it false?

propi62, 2024-09-05:
my switches should never turn off. i had alot of trouble because my switches turned off whene updating.
(whole It equippment turned off while working, washing machine turned off while washing)
i hope this will not happen in future updates.

jpettitt, 2024-09-14:
It should default to false like it used to.  This was a breaking change and should not have been pushed without notification.

TommySharpNZ, 2024-09-19:
> > my switches should never turn off. i had alot of trouble because my switches turned off whene updating. (whole It equippment turned off while working, washing machine turned off while washing) i hope this will not happen in future updates.
> 
> For this problem, we are about to release a plug without relay version for EU, which can solve this problem physically.

PLEASE also release that for AU/NZ and also a bit skinnier to allow 2 plugs in one double socket.... Please, I have been hunting for a power meter only plug without relay.

Roving-Ronin, 2024-11-12:
@tarontop 
Hi Aiden,  any update as to the 'no relay' plugs for AU becoming available? Thought they should have come out earlier this month? 
With 2 more dead plugs (on critical equipment) really looking for these to be released asap (if only on esp32).

Thanks