athom-tech/athom-configs #75: Smart Plug v3 still using esp8266 despite having an esp32-c3 board?

alucryd, 2024-09-03:
Hi there, any reason to stick with esp8266 on the v3 smart plugs despite having an esp32-c3 board inside? Could you share the exact board you are using? I'd like to run some power consumption tests between the 2 configs, as well as give esp-idf a go.

Roving-Ronin, 2024-09-04:
Smart Plug v1 and v2 use the esp8285 chip, whilst the v3 uses the esp32-c3 chip. 

No sure what you mean by 'stick with' as the v3 has the C3 soldered to its PCB, same as the v1/v2 has the 8285 soldered on. Not like you have an option to swap it (why would you want to anyhow).

Software wise not like you have an option either, as when compiling you have to define which chip its for, as you can't install 8285 on ESP32 (or vice versa).

If you want to play around with the code then fork the repositories, the v2 is at: https://github.com/athom-tech/athom-configs/blob/main/athom-smart-plug-v2.yaml and the v3 is at: https://github.com/athom-tech/esp32-configs/blob/main/athom-smart-plug.yaml

alucryd, 2024-09-04:
Thank you for the links, I was in fact looking at the wrong repository and was referencing the config file for the v1 plug instead of the v3 (I only have v3 plugs). Apologies for the noise.

closed 2024-09-04 by alucryd