athom-tech/athom-configs #77: Athom Smartplugs V2 & V3 no longer display measurements in webserver page since ESPHome 2024.8.1 update

effevee, 2024-09-07:
The smartplugs work ok : sensor values and diagnostics are shown correctly in HA Settings and on dashboard. The webserver display however fails to show the sensor messages. Only these warnings are shown 

Time	level	Tag	Message
11:36:07	[W]	[component:237] Component preferences took a long time for an operation (65 ms).
11:36:07	[W]	[component:238] Components should block for at most 30 ms.
11:41:08	[W]	[component:237] Component preferences took a long time for an operation (62 ms).
11:41:08	[W]	[component:238] Components should block for at most 30 ms.
11:46:08	[W]	[component:237] Component preferences took a long time for an operation (62 ms).
11:46:08	[W]	[component:238] Components should block for at most 30 ms.

HA ESPHome logs don't show any errors. What could be wrong ? Is this because the sensor fails to respond within a certain timeframe?  If yes, is there anything that can be added to the yaml config to make this work again ?

Roving-Ronin, 2024-09-07:
G he age you set log_level to DEBUG?

Roving-Ronin, 2024-09-07:
The component time error is a known issue, the last two lines in this log setup gets rid of them.

logger:
  level: ${log_level}
  baud_rate: 0
  logs:
    component: ERROR # Fix for issue #4717 "Component xxxxxx took a long time for an operation"

effevee, 2024-09-08:
> G he age you set log_level to DEBUG, if you want the sensor messages every 10sec?

after changing loglevel to DEBUG, the sensor messages are back. Thanks :1st_place_medal:

closed 2024-09-08 by effevee