athom-tech/athom-configs #79: Invalid parity: Integration requested parity NONE but you have EVEN!

mazay, 2024-09-11:
Newly purchased V3 plugs are throwing errors with ESPHome version 2024.8.3:
```
[22:26:29][I][app:100]: ESPHome version 2024.8.3 compiled on Sep 10 2024, 15:41:08
[22:26:29][I][app:102]: Project Athom_Technology.Smart_Plug_V3 version v1.0.3
[22:26:29][E][uart:029]:   Invalid parity: Integration requested parity NONE but you have EVEN!
[22:26:39][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
```

As far as I can tell the `uart` block is missing in V3 config. Please advise what it should look like.

V2 block for reference - https://github.com/athom-tech/athom-configs/blob/main/athom-smart-plug-v2.yaml#L111-L113

Roving-Ronin, 2024-09-11:
v3 has the UART code:

```
uart:
  rx_pin: GPIO20
  baud_rate: 4800
  data_bits: 8
  stop_bits: 1
  parity: EVEN
```
at: https://github.com/athom-tech/esp32-configs/blob/e4912eb6e107509176f5e6d5b25c9aec9ff16639/athom-smart-plug.yaml#L99

Also worth checking closed issues first:

https://github.com/athom-tech/esp32-configs/issues/2

https://github.com/esphome/issues/issues/4529#issuecomment-1822283401

mazay, 2024-09-12:
not sure how I missed closed issues, they did not appear in the search results for me...

also, I didn't realise the v3 config was in another repo

so it's about `parity` settings then, should it be updated?

mazay, 2024-09-16:
according to [this](https://github.com/athom-tech/esp32-configs/issues/9) the setting is correct

closing anyway as it does not belong to this repo

closed 2024-09-16 by mazay