athom-tech/athom-configs #81: Timezone of smart plug incorrect

TommySharpNZ, 2024-09-17:
I've adopted my smart plug into my ESPHome environment and added a simple sensor to show the current date/time.
As I suspected it's incorrect even though the timezone in my Home Assistant instance is set correctly to be GMT+13 for New Zealand.
Any thoughts on what I can do to resolve this? 
Right now the "Daily" sensors are not resetting at the correct time :-(

Roving-Ronin, 2024-09-17:
Have you looked at the yaml file and the substitutions, namely the Timezone being one that you add to your devices local yaml file and enter in the TZ for your area I.e. "Pacific/Auckland".

TommySharpNZ, 2024-09-17:
The standard YAML states the timezone is used if the device is in a different timezone to the HA server. So I have left it blank.
I have just added the below setting to my local YAML file which I'm glad to say seems to "override" that setting in the linked yaml package and now my time is showing as correct.

time:
  - platform: homeassistant
    id: homeassistant_time

Roving-Ronin, 2024-09-21:
Simpler way is as per yaml file notes. In the substitutions section have this:

```
substitutions:
  # Current Limit in Amps. AU Plug = 10. IL, BR, EU, UK, US Plug = 16.
  current_limit : "10"
  # Set timezone of the smart plug. Useful if the plug is in a location different to the HA server. Can be entered in unix Country/Area format (i.e. "Australia/Sydney")
  timezone: "Pacific/Auckland"
```

I would assume NZ is the same as AU in regards current limit on a GPO, so you'll also want to change the current_limit to "10".

I'd suggest looking at the yaml for the v2 / v3 plugs and reading the comments, as you need to edit this to update it to have a name specific to your needs, give it a room location (that HA uses to auto populate the room/zone its in) etc.
Using SNTP, instead of legacy HA, also allows the plugs to operate independant of using HA or it being unavailable.

v2 plug - https://github.com/athom-tech/athom-configs/blob/main/athom-smart-plug-v2.yaml

v3 plug - https://github.com/athom-tech/esp32-configs/blob/main/athom-smart-plug.yaml