athom-tech/athom-configs #82: lost total daily energy sensor in v2 plug

rich-kettlewell, 2024-09-18:
since updating to the latest esphome and reinstalling the latest athom-smart-plug-v2.yaml script, i’ve lost my total_daily_energy sensors.

endyb66, 2024-09-25:
Right. The icon with the "24" number makes a lot of sense to me, the name "Total Energy Since Boot" however not so...

Roving-Ronin, 2024-09-25:
Iirc the plugs have one reading that is supposed to be the 'daily' reading (but it's crap as if you reboot or flash it goes back to zero also) and another that is the Total sensor (kind of like a lifetime total).

IMHO best option was to ditch them all and move to using a custom component that gives a daily, yesterday, weekly, monthly and yearly sensor.

[image]

endyb66, 2024-09-25:
What's the custom component you are using?

Roving-Ronin, 2024-09-26:
Have a look at the configuration that I run: 

v2 Plugs - https://github.com/Roving-Ronin/myHomeAssistant/blob/main/esphome/athom-smart-plug-v2.yaml

v3 Plugs - https://github.com/Roving-Ronin/myHomeAssistant/blob/main/esphome/athom-smart-plug-v3.yaml

Note that these configs call other yaml to then compile the 'full' yaml for each device, namely these plugs call:

Base Config, that applies the same set of standard sensors that ALL my esphome devices have:
  https://github.com/Roving-Ronin/myHomeAssistant/blob/main/esphome/common/base-configuration.yaml)

WiFi Config, sets up WiFi and adds the WiFi sensors I use:
  https://github.com/Roving-Ronin/myHomeAssistant/blob/main/esphome/common/network-wifi.yaml

Athom Plug config, allows having one file with all the athom plug specific settings in it, instead of seperate per version to maintain:
  https://github.com/Roving-Ronin/myHomeAssistant/blob/main/esphome/sensors/athom-power-plugs.yaml

In the athom plug config, I call the custom component (based off Dentra's, but expanded) using:

```
external_components:
  - source: github://roving-ronin/myhomeassistant/components@main
    refresh: 5s
```

Then from line 193 to 233 I call the sensors this adds: https://github.com/Roving-Ronin/myHomeAssistant/blob/cfbc444b6f77b36c1b5101c47481f049117d9b0b/esphome/sensors/athom-power-plugs.yaml#L193

I keep the existing Total Energy sensor (that roughly shows the lifetime kWh..or at least to some reflash stuffs it) at: https://github.com/Roving-Ronin/myHomeAssistant/blob/cfbc444b6f77b36c1b5101c47481f049117d9b0b/esphome/sensors/athom-power-plugs.yaml#L193

I also keep the existing Energy sensor (that I think you refer to as the 24hr one, but its useless as it resets when rebooted / flashed etc), but just change it to be 'internal', so it works within the ESP device and other sensors can call it if needed, but otherwise it isn't visible on the web GUI or published to HA: https://github.com/Roving-Ronin/myHomeAssistant/blob/cfbc444b6f77b36c1b5101c47481f049117d9b0b/esphome/sensors/athom-power-plugs.yaml#L185

I also use webserver v3 (instead of v2 your used to seeing) that allows sorting the order of sensors in the WebGUI (not HA) and if you click on a sensor it will show a graph of its data. 

Oh and in my Base config I also run an extra text sensor that allows defining the 'Power Circuit' (I have 3 phase and 3 different circuits for wall outlets in the house), that allows filtering ESP devices by the circuit they're on / turning devices off if they are on a certain phase etc.

I have to update my template for one thing currently, one I sort out this issue: https://github.com/athom-tech/athom-configs/discussions/87  pending that you can just apply it and make sure to update the selector in 'Power on State' to what you want.

Web Server GUI looks like:

[image]

Note: 

1. The new on_boot selector that athom added, even though I've changed to '2' (ALWAYS_ON) seems to come up with 'Power On State' is 'Always_Off' and the actual relay/swith turned off. You need to go to the WebGUI and turn on the relay (aka GPO Switch) and change the 'Power On State' to 'ALWAYS ON' (if thats your preference).

2.  The Week / Month / Year sensors will only start counting from the following Monday 0:00am / 1st of Month / 1st day of Year)

3. I've adding the mdi icon for the 'select' switch that has been recently added in athom repo, that MAY cause the core image to be rebuilt that then causes the periodical energy sensors to default to 0.0 kWh / NA. To avoid this I've just added the functionality into the custom component for it to store the value (in the ESP memory) and restore it by default. This is like adding 'globals' and then having the sensors write to them, but adding this functionality by default within the custom component. 

Could I ask that if you use this component, you let me know if this is working when booting/flashing in future. It should but feedback/confirmation appreciated.

Here's what the WebGUI should look like (yes the periodical energy sensors are NA....as adding that function (No.3 above) I had to take a hit for the team and reflash mine and see them reset to 0.0)

[image]

Roving-Ronin, 2024-09-26:
If you want to use the above, you can either FORK my repository and pull off your own repository, or else here's the yaml I use for a device to work in my ESPHome/HA and pull the code for the above.  

**_Please remember to update the substitutions to reflect the names / HA zone location / Power Circuit of your own device(s)._**

**Smart Plug v2**
```
substitutions:
  # Device name   
  name: "gpo-ups-study"
  # Device friendly name 
  friendly_name: "UPS - Study"
  # Description as appears in ESPHome & top of webserver page
  device_description: "Athom Plug v2 - Phase C / GPO 1 - Study PCs & Monitors"
  # Allows ESP device to be automatically lined to an 'Area' in Home Assistant. Typically used for areas such as 'Lounge Room', 'Kitchen' etc    
  location: "Study"
  # The phase in the home power supply, upon which this plug is utilsed. 1p = A. 3p = A, B or C.  (myHome GPO 1 = Phase C, GPO 2 = Phase A, GPO 3 = Phase B)
  power_circuit: "TBC"

esphome:
  name_add_mac_suffix: false

esp8266:
  board: esp8285
  restore_from_flash: true

packages:
  remote_package:
    url: https://github.com/Roving-Ronin/myHomeAssistant/
    ref: main 
    files: [esphome/athom-smart-plug-v2.yaml]
    refresh: 5d # optional
```

**Smart Plug v3**
```
substitutions:
  # Device name
  name: "gpo-ups-network"
  # Device friendly name
  friendly_name: "UPS - Network Rack"
  # Description as appears in ESPHome & top of webserver page
  device_description: "Athom Plug v3 - Phase C / GPO 1 - Network Rack Devices"
  # Allows ESP device to be automatically lined to an 'Area' in Home Assistant. Typically used for areas such as 'Lounge Room', 'Kitchen' etc
  location: "Network Rack"
  # The phase in the home power supply, upon which this plug is utilsed. 1p = A. 3p = A, B or C.  (myHome GPO1 = Phase C, GPO2 = Phase A, GPO3 = Phase B)
  power_circuit: "GPO 2"  

esphome:
  name_add_mac_suffix: false

packages:
  remote_package:
    url: https://github.com/Roving-Ronin/myHomeAssistant/
    ref: main 
    files: [esphome/athom-smart-plug-v3.yaml]
    refresh: 5d # optional
```

endyb66, 2024-09-26:
Thank you so much for this detailed response! This answers almost all of my questions. I think I'll stick to the original yaml for the moment, I just need to adjust the new "Total Energy Since Boot" entity, although I don't really use / need it anyway.
I'm sure your answer does help others with the same issues and questions!

Roving-Ronin, 2024-10-03:
If of any interest to people, I've added in:

- The ability to have the component automatically save the periodical readings (i.e. daily, weekly etc) to NVS flash memory every 5 minutes. This is done without having to define globals or sensors to do it, the custom component just automatically does it. 

If you want to make this time period longer, run off your own fork and just change the 300 (5min in seconds) into something longer, at: https://github.com/Roving-Ronin/myHomeAssistant/blob/4bb6a50061d496e6e782849109831cdd588173db/components/energy_statistics/energy_statistics.h#L33.  PS. If you do this you should also change the following line in the athom yaml to match, so the Energy Total defined in globals is saving at the same frequency:

```
preferences:
  flash_write_interval: 5min
```

Can't work out how to get it to work properly atm, but would like to see if I can add a number input (or pull down with 5min / 15min / 30min etc options) that displays on the WebGUI/HA, that the component would then read and dynamically change the save period.

- Added the ability to reset the periodical (daily / weekly etc) energy readings back to 0.0 kWh. This leaves the 'Energy Total' value stored in flash though, in case you want to track that, but reset the periodicals. If wanting to erase them all, including the 'Energy Total' then just hit the Factory Reset.

- Cosmetic change, instead of the readings show 'NA', pending their reaching the start of a new cycle (i.e. Monday 00:00am for week, 1st of Month, 1st day of year), these will show "0.000 kWh".

BTW there's other sensors to do the same for Water (L) and Gas in m3 and megajoules (MJ) - see screenshot below. If anyone wants to test, it would be appreciated. Note: As gas / water usage and water readings aren't as frequent compared to power, these run with a 15min time period between saving to ESP flash / lifespan.

Next I have to see about:

- Adding a check to see if the water / gas values have updated since last changed and if they haven't then skip saving to flash (to reduce flash writes / increase life of device)

- Making a 'utility_statistics' component, that merges the gas and water periodicals into one component, for when your running both on the same ESP device. This will allow both water and gas readings to be saved at the same time, thereby also reducing flash writes.

PS. Was we have mandatory rain water tanks, that are used for things like laundry / garden water, the 'normal' Water above is labelled "Town Water", as I intend to add a "Tank Water" in future (with a pulse meter added into the supply from rain water tank), so it will also then show daily/weekly etc for the Rain/Tank Water, and probably then have a 'Water' sensor that shows shows combined usage for all water types in total, plus daily/weekly etc.

If this is of use to people please shout out, wondering if this is something that others see value in and if it should be submitted as a built in component for ESPHome?

[image]