athom-tech/athom-configs #90: No Power or current reported since recent upgrade to ESPhome on V2 plugs

atmezferix, 2024-10-02:
No Power or Current reported and no logs displayed on the web interface other than the following messages:-

Time | level | Tag | Message

20:08:05 | [I] | [safe_mode:041] | Boot seems successful; resetting boot loop counter
20:08:05 | [W] | [component:237] | Component safe_mode took a long time for an operation (78 ms).
20:08:05 | [W] | [component:238] | Components should block for at most 30 ms.
20:11:01 | [W] | [component:237] | Component preferences took a long time for an operation (74 ms).
20:11:01 | [W] | [component:238] | Components should block for at most 30 ms.
20:16:01 | [W] | [component:237] | Component preferences took a long time for an operation (77 ms).
20:16:01 | [W] | [component:238] | Components should block for at most 30 ms.

Lost all my energy monitoring data.

Switch still operates and Voltage is displayed on web interface but nothing in the logs.

Older V1 plugs are still working fine.

Roving-Ronin, 2024-10-03:
Have you changed log_level: "INFO" to log_level: "DEBUG" and reinstalled the firmware, if wanting to really check what is being logged?

Are you in Safe Mode, that would disable all components (and hence Power/Energy monitoring)? If so, try reboot and/or reset to factory.

Just updated a v2 to ESPHome 2024.9.2 and works fine.

atmezferix, 2024-10-03:
Thanks it was the log level.

I added

logger:
  level: DEBUG

And I can see the updates now.

Thanks for your help.

Roving-Ronin, 2024-10-04:
No need to "Add" it into the main body of the yaml. The way the file is configured is so that you can made these changes (to suit individual needs) within the individual devices 'Substitutions' Section at the top. 

For example here's the full substitutions section for the v2 plug (v1 and v3 should be the same also) from the GitHub yaml that is pulled down when compiling:

```
substitutions:
  name: "athom-smart-plug-v2"
  friendly_name: "Smart Plug V2"
  # Allows ESP device to be automatically lined to an 'Area' in Home Assistant. Typically used for areas such as 'Lounge Room', 'Kitchen' etc
  room: ""
  device_description: "athom smart plug v2"
  project_name: "Athom Technology.Smart Plug V2"
  project_version: "v2.0.5"
  sensor_update_interval: "10s"
  relay_restore_mode: RESTORE_DEFAULT_ON
  # Current Limit in Amps. AU Plug = 10. IL, BR, EU, UK, US Plug = 16.
  current_limit : "16"
  # Define a domain for this device to use. i.e. iot.home.lan (so device will appear as athom-smart-plug-v2.iot.home.lan in DNS/DHCP logs)
  dns_domain: ""
  # Set timezone of the smart plug. Useful if the plug is in a location different to the HA server. Can be entered in unix Country/Area format (i.e. "Australia/Sydney")
  timezone: ""
  # Set the duration between the sntp service polling ntp.org servers for an update
  sntp_update_interval: "6h"
  # Network time servers for your region, enter from lowest to highest priority. To use local servers update as per zones or countries at: https://www.ntppool.org/zone/@
  sntp_server_1: "0.pool.ntp.org"
  sntp_server_2: "1.pool.ntp.org"
  sntp_server_3: "2.pool.ntp.org"  
  # Enables faster network connections, with last connected SSID being connected to and no full scan for SSID being undertaken
  wifi_fast_connect: "false"
  # Define logging level: NONE, ERROR, WARN, INFO, DEBUG (Default), VERBOSE, VERY_VERBOSE
  log_level: "INFO"
  # Enable or disable the use of IPv6 networking on the device
  ipv6_enable: "false"
  # Hide the ENERGY sensor that shows kWh consumed, but with no time period associated with it. Resets when device restarted and reflashed.
  hide_energy_sensor: "false"
  # Specifies whether pins should be initialised as early as possible to known values. Recommended value is false where switches are involved, as these will toggle when updating the firmware or when restarting the device. Defaults to true.
  early_pin_init: "true"
```

To change any setting in that section, all you need to do is add the substitution name and the state you want it to be, to your local device config. For example here's one for a plug running on a UPS in the Study. If you look at the template on Github is shows you the options that can be set. So to add debug log to a plug, you'd add it to the local device config, as per below:

```
substitutions:
  # Device name   
  name: "gpo-ups-study"
  # Device friendly name 
  friendly_name: "UPS - Study"
  # Description as appears in ESPHome & top of webserver page
  device_description: "Athom Plug v2 - Phase C / GPO 1 - Study PCs & Monitors"
  # Allows ESP device to be automatically lined to an 'Area' in Home Assistant. Typically used for areas such as 'Lounge Room', 'Kitchen' etc    
  location: "Study"
  # Define logging level: NONE, ERROR, WARN, INFO, DEBUG (Default), VERBOSE, VERY_VERBOSE
  log_level: "DEBUG"

```

This the means your local devices are still kept in alignment with any updates to the Github template, but allows your device to then have its own specific settings that are applied over the top of the Github template.

Roving-Ronin, 2025-01-09:
@atmezferix could you please close this issue, as it was resolved back on 4th Oct 2024.