athom-tech/athom-configs #94: ESPHome 2024.10 not working with configuration

axute, 2024-10-16:
Hello, 

I updated my esphome to the latest version. 
Now the update for the device is not working anymore. 
Could you please update your examples? 

```
INFO ESPHome 2024.10.0
INFO Reading configuration /config/esphome/asp2-1.yaml...
Failed config

uart: [source /config/esphome/asp2-1.yaml:**]
  - rx_pin: 
      number: 3
      mode: 
        input: True
        output: False
        open_drain: False
        pullup: False
        pulldown: False
        analog: False
      inverted: False
    baud_rate: 4800
    rx_buffer_size: 256
    stop_bits: 1
    data_bits: 8
    
    Component cse7766 requires parity EVEN for the uart referenced by uart_id.
    parity: NONE
```

propi62, 2024-10-16:
same problem
[image]

endyb66, 2024-10-17:
> to update i had to add uart:
> 
> ```
> substitutions:
>   name: strom-aquarium
>   friendly_name: strom_aquarium
> packages:
>   Athom_Technology.Smart_Plug_V2: github://athom-tech/athom-configs/athom-smart-plug-v2.yaml
> esphome:
>   name: ${name}
>   name_add_mac_suffix: false
>   friendly_name: ${friendly_name}
> 
> wifi:
>   ssid: !secret wifi_ssid
>   password: !secret wifi_password
> 
> uart:
>   rx_pin: RX
>   baud_rate: 4800
>   parity: EVEN
> ```

Would be great if someone could explain this a bit more.