athom-tech/athom-configs #101: Upgrading to ESPHome 2024.10.0 prevents OTA from working

leonardospina, 2024-10-18:
I have a few athom-4gang-switch devices around the house, and after the latest ESPHome update I installed this morning I cannot do OTA anymore, apparently the binary file is too large for OTA.
Updating form the ESPHome integration fails because of issues with name resolution (just ignore it), posting just to show the logs that show the size of the binary:

[image]

but if I download the binary and upload it via the device's web UI I get this:

[image]

madberry, 2024-10-18:
Have you defined what esp is being used. I've seen this error come up a few times and after defining the esp OTA started to work again.

leonardospina, 2024-10-18:
the board type is defined in the config in the original Athom's repo:

[image]

madberry, 2024-10-18:
Have you tried to reboot the switch and then attempt the flash again?

leonardospina, 2024-10-18:
No I haven't, I mean that's hardwired to the mains not really a quick thing... one thing I might try is resetting it to factory defaults and then flash it, but the strange thing is that it's not just one, all of them (same device type) have the same issue since I upgraded ESPHome

leonardospina, 2024-10-18:
yeah very annoying... I'll try with one anyway and see if it works...

Jannis87, 2024-10-18:
its an issue for me too. 

i think the binary is to big:
Flash: [=====     ]  53.9% (used 552101 bytes from 1023984 bytes)

leonardospina, 2024-10-19:
yep same here, I have many devices and only the switches have that problem

Jannis87, 2024-10-19:
> yep same here, I have many devices and only the switches have that problem

gzip -9 firmware.bin worked for me :). The compressed file can be uploaded via the web interface. Its a workaround, but at least my devices are up to date :)

madberry, 2024-10-19:
I posted this issue to the esphome discord hopefully they will be able to pick it up.

leonardospina, 2024-10-21:
Thanks for the hint I'll give it a try!

leonardospina, 2024-10-21:
wow it worked! thanks @madberry

cloudlena, 2025-03-20:
My solution is to have a very minimal configuration ready in ESPHome and always first flash that before I do any update. The minimal config results in a small firmware so that there is enough space on the device to then upload the actual update.

It's not nice, but it works 😅

nirev, 2025-07-30:
@cloudlena can you share your minimal config? : )

MaJerle, 2025-07-30:
> [@cloudlena](https://github.com/cloudlena) can you share your minimal config? : )

That's the strategy I can propose you for the product.

https://github.com/athom-tech/athom-configs/issues/122#issuecomment-3057866563

nirev, 2025-07-30:
@MaJerle thank you! That worked perfectly : )

cloudlena, 2025-08-03:
> [@cloudlena](https://github.com/cloudlena) can you share your minimal config? : )

```yaml
substitutions:
  name: athom-presence-sensor-abc123
  friendly_name: Athom Presence Sensor abc123

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}

api:
  encryption:
    key: XXX

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

esp8266:
  board: esp8285
  restore_from_flash: true
ota:
  - platform: esphome
```