athom-tech/athom-configs #103: smart v2 plug reactive power warning

bucky2076, 2024-10-20:
this annoying warning coming out in logs for all athom v2 plugs...

>
Entity sensor.athom_smart_plug_v2_a77732_reactive_power (<class 'homeassistant.components.esphome.sensor.EsphomeSensor'>) is using native unit of measurement 'VAR' which is not a valid unit for the device class ('reactive_power') it is using; expected one of ['var']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+esphome%22
>

Roving-Ronin, 2024-10-21:
This issue is as HA is expecting 'var' (officially the correct version), whilst ESPHome expects 'VAR' (also used in the world, but not 'official').

See: https://github.com/esphome/esphome/pull/7643

Until this is officially fixed, suggest you add the unit into your local device yaml, like this:

```
    reactive_power:
      name: "Reactive Power"
      unit_of_measurement: var
      accuracy_decimals: 0   
```

Roving-Ronin, 2024-11-14:
@bucky2076 looks like this will be 'fixed' in ESPHome 2024.11.0, as the above PR pull request was just approved for this forthcoming release. This will see 'VAR' in ESPHome changed to use 'var', as Home Assistant is expecting and is the (official) standard way of referring to this globally.

Roving-Ronin, 2024-11-21:
@bucky2076 ESPHome 2024.11.0 is released and does include an update to fix this issue (changing from VAR to var).

Roving-Ronin, 2024-12-05:
@bucky2076 can you close this issue please.

closed 2024-12-05 by bucky2076