athom-tech/athom-configs #105: No Relay Power Monitoring EU Plug for ESPHome links to wrong yaml

Gunni, 2024-11-06:
Hey,

The [No Relay Power Monitoring EU Plug for ESPHome](https://www.athom.tech/blank-1/no-relay-consumption-monitoring-eu-plug-for-esphome) links to [wrong yaml](https://github.com/athom-tech/athom-configs/blob/main/athom-smart-plug-v2.yaml).

You need a new yaml without the switch part. Or any of the power switch logic.

Roving-Ronin, 2024-11-07:
@Gunni 
I'm still waiting for this plug to come out in the AU version, so can't test it myself. If you can...in your local device file, try adding:

```
select:
  - platform: template
    name: "Power On State"
    optimistic: true
    icon: "mdi:electric-switch"
    options:
      - Always Off
      - Always On
      - Restore Power Off State
    internal: true

switch:
  - platform: gpio
    name: "Switch"
    pin: GPIO15
    internal: true

```
then validate and flash it to your 'no relay' plug.  If you could confirm this does what your after, that would appear to be to remove the 'Power On State' and 'Switch' options from the devices webGUI / HA? 

Could you confirm this and also that no other errors are seen? The above should hide these, whilst the switch logic should just lie dormant in the background.

If this 'solves the issue' then we can see about making this into a subsitute option, so users of these plugs could add something like:

```
is_no_relay_plug: 'true'
```

to their substitutions section and have the existing Smart Plug v2 code work for them.

Gunni, 2024-11-11:
I don't have one yet, it is out of stock for me (EU). I just noticed that the yaml had on/off stuff despite that device specifically not having that capability, which is what I want.

Roving-Ronin, 2024-11-13:
@tarontop keeping with your naming standard to date, would this not be better as either:
a) athom-smart-plug-v2-without-relay.yaml , or:
b) athom-smart-plug-without-relay.yaml

Gunni, 2024-11-23:
Weirdly the product is gone from the website and a ban on shipments to the EU on the website...

What's happening?

Roving-Ronin, 2024-12-05:
@Gunni  Legal BS. https://community.homey.app/t/athom-wins-lawsuit-against-chinese-athom-tech-and-aliexpress/121124

Gunni, 2024-12-05:
But why remove it entirely? Iceland is not in the European Union, yet.

Gunni, 2025-01-03:
Still wrong link though.

Roving-Ronin, 2025-01-03:
@athom-tech / @tarontop 

Aiden, What @Gunni is referring to is that the URL on the Athom WEBSITE (not Github) is telling 'No Relay' users to go to the v2 smart plug yaml and NOT the new one of the No Relay plug.

from: https://www.athom.tech/blank-1/no-relay-consumption-monitoring-eu-plug-for-esphome

[image]

The URL on the above page of the Athom website needs to be corrected to: https://github.com/athom-tech/athom-configs/blob/main/athom-without-relay-plug.yaml