athom-tech/athom-configs #111: Missing .local in domain name for Athom devices

rmaeder, 2024-12-04:
When I adopt an Athom Smart Plug V2 in ESPHome,
The link under the "VISIT" button uses the domain name athom-smart-plug-v2-ecXXX without .local appended; this does not work. Other devices, such as an Apollo Multisensor Mk1 use a domain name of apollo-msr-1-XXXX.local, which usually works.

When trying to update it, there is this error after the compilation, when it wants to upload the new configuration:

INFO resolving IP address of athom-smart-plug-v2-ecXXX

which fails for the same reason: the .local suffix is missing.

workaround: in the esphome config file for the device, add `domain: ".local"`:
```
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: ".local"
```
this should probably happen here: https://github.com/athom-tech/athom-configs/blob/main/athom-smart-plug-v2.yaml#L14

Roving-Ronin, 2024-12-05:
@rmaeder It should be working fine with nothing in that substitution field, as it defaults to .local that way. If you unit isn't working then for YOUR devices, in each individual devices configuration file in ESPHome interface, you should add this:
```
substitutions:
  dns_domain: ".local"
```

For example your config for each device should look like:

```
substitutions:
  name: "gpo-clothes-dryer"
  friendly_name: "Clothes Dryer"
  device_description: "Athom Plug v2 - Phase B / GPO 3 - Clothes Dryer"
  location: "Laundry"
  dns_domain: ".local"

esphome:
  name_add_mac_suffix: false

esp8266:
  board: esp8285
  restore_from_flash: true

packages:
  remote_package:
    url: https://github.com/Roving-Ronin/myHomeAssistant/
    ref: main 
    files: [esphome/athom-smart-plug-v2.yaml]
    refresh: 5s # optional
```

(ignore the remote_package section, that points to my personal version for these plugs... yours should point to the https://github.com/athom-tech/athom-configs/blob/main/athom-without-relay-plug.yaml instead).

Bert-R, 2024-12-05:
I faced the same problem as @rmaeder. In my case, I use a domain suffix at home (e.g. `home.my.domain`) and because of that, I the "Update all" option of ESP Home didn't work for my Athom smart plugs until I explicitly included the domain suffix in the plug configuration. I don't have to do that for any of my other ESP Home devices, so I agree with Roman that it needs improvement in the default configuration.

Roving-Ronin, 2024-12-05:
@rmaeder  

@Bert-R   assume your using .home.my.domain  (needs the . at the start).  Scratching my head as this setting is used in (essentially) every Athom products yaml configuration, yet this is the first issue raised saying it isn't defaulting to using .local ?

From https://esphome.io/components/wifi.html it says:
```
domain (Optional, string): Set the domain of the node hostname used for uploading. For example, if it’s set to .local, all uploads will be sent to <HOSTNAME>.local. Defaults to .local.
```

with substitutions  dns_domain: "" being blank/empty it should be default to .local
What version of ESPHome are you both using? Has this just started / old issue (i.e. started with a recent ESPHome update)?

Bert-R, 2024-12-05:
Yes, it is configured as:
```
substitutions:
  dns_domain: ".home.my.domain"
```

The issue is that I need to do that for the Athom plugs only. For my other ESP Home devices I do not have to specify the domain suffix defined.

The issue came up after an update of ESP Home. At that time, I reported it here: https://github.com/esphome/issues/issues/5827#issuecomment-2198320513

rmaeder, 2024-12-05:
> with substitutions dns_domain: "" being blank/empty it should be default to .local
> What version of ESPHome are you both using? Has this just started / old issue (i.e. started with a recent ESPHome update)?

I've always had this issue (missing domain .local) ever since I adopted two of these plugs into ESPHome about two months ago. Currently I am on ESPHome 2024.11.2

I notice that in the device source, at line 14, https://github.com/athom-tech/athom-configs/blob/main/athom-smart-plug-v2.yaml#L14
it explicitly sets the dns_domain to "" in your substitutions. I don't see this in other devices (Everything presence lite, Apollo MSR-1).
Doesn't this override the built-in default of ".local"?

I changed my original workaround to your suggestion,
```
substitutions:
  dns_domain: ".local"
  ...
```
and it does work.

Roving-Ronin, 2024-12-31:
@rmaeder If this was confirmed as solving the issue, was there a PR raised to update all the products template yamls?

Roving-Ronin, 2025-01-09:
@rmaeder could you please close this issue, as .local has now been added as standard into the yaml

rmaeder, 2025-01-09:
Seems to be working.

closed 2025-01-09 by rmaeder