athom-tech/athom-configs #114: Updates fail with "Error binary size: Unknown error from ESP"

rmaeder, 2024-12-21:
ever since ESPHOME 2025.12.0, attempts to update my Athom smart Plugs V2 fail with these messages;
```
INFO ESPHome 2024.12.1
INFO Reading configuration /config/esphome/athom-smart-plug-v2-ecXXXX.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing athom-smart-plug-v2-ecXXXX (board: esp8285; framework: arduino; platform: platformio/espressif8266@4.2.1)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.6
|-- ArduinoJson @ 6.18.5
RAM:   [=====     ]  46.5% (used 38056 bytes from 81920 bytes)
Flash: [======    ]  58.1% (used 595041 bytes from 1023984 bytes)
========================= [SUCCESS] Took 2.98 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.XXX.YYY port 8266...
INFO Connected to 192.168.XXX.YYY
INFO Uploading /data/build/athom-smart-plug-v2-ecXXXX/.pioenvs/athom-smart-plug-v2-ecXXXX/firmware.bin (599200 bytes)
INFO Compressed to 408955 bytes
ERROR Error binary size: Unknown error from ESP
```

rmaeder, 2024-12-24:
When I do a "Clean build files" first, the update works.
Immediately doing a second update without cleaning, fails as above.
So, perhaps this is a problem with the dependencies and there is some junk lying around.

GazTheGeek, 2025-05-12:
I have this issue too. Is it possible to have a revision to the yaml that compiles to a smaller bin?