athom-tech/athom-configs #131: athom without relay will not compile with esphome 2025.9.0

dmshimself, 2025-09-17:
Compiling an 'athom without relay' gives the following errors in esphome 2025.9.0  (Previous version worked fine).  I've cleaned the installation files and rebuilt, but that doesn't help.  Other esphome devices such as the athom 4-channel relay compile just fine.

> 

In file included from /root/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266mDNS/src/ESP8266mDNS.h:48,
                 from /root/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266mDNS/src/ESP8266mDNS.cpp:23:
/root/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266mDNS/src/LEAmDNS.h:116:10: fatal error: WiFiUdp.h: No such file or directory

*****************************************************************
* Looking for WiFiUdp.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:WiFiUdp.h"
* Web  > https://registry.platformio.org/search?q=header:WiFiUdp.h
*
*****************************************************************

  116 | #include "WiFiUdp.h"
      |          ^~~~~~~~~~~
compilation terminated.
*** [.pioenvs/athom-without-relay-plug-5755de/lib0d3/ESP8266mDNS/ESP8266mDNS.cpp.o] Error 1

BeebleZap, 2025-09-17:
I also have same error for athom-smart-plug-v2

effevee, 2025-09-17:
> It's an issue in ESPHome, see [esphome/esphome#10770](https://github.com/esphome/esphome/issues/10770)

I can confirm it is indeed a problem with mdns in this esphome 2025.9.0 version. If you disable mdns in the yaml file, it compiles ok. 

BUT without mdns you will have to add manual ip settings to your yaml file or HA will not find your device anymore !!!!!

```
# esphome 2025.9.0 fatal error due to mdns ?
mdns:
  disabled: true

wifi:
  manual_ip:
    static_ip: 192.168.1.107
    gateway: 192.168.1.1
    subnet: 255.255.255.0
    dns1: 192.168.1.15
    dns2: 192.168.1.16

  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: !secret domain

```

MaJerle, 2025-09-17:
Better to wait hot fix (it was fixed and will be released soon) and then update.

hiraniha, 2025-09-18:
I have the same issue when using this package 

github://athom-tech/athom-configs/athom-presence-sensor.yaml

Motya-HA, 2025-09-18:
@MaJerle  - define "soon", please

thgoncalves, 2025-09-19:
> ESPHome 25.9.1 has just been released with the fix; No more compilation problems :-)

Can confirm.. all good now